SuLingGG/OpenWrt-Docker #5: The rpi 3/3b image has got wrong architecture

bookyue, 2020-10-19:
# Problem
As the issue title says, it should be arm arch. But now it's amd64 arch out there.
```
root@DietPi:~# docker image inspect sulinggg/openwrt:rpi3
[
    {
        "Id": "sha256:a304ad969f09d42c62b5aefa603d3015a803748c63debd2c41a0304bffe9f0d6",
        "RepoTags": [
            "sulinggg/openwrt:rpi3"
        ],
        "RepoDigests": [
            "sulinggg/openwrt@sha256:c66ff3b3f7984c224b8f542683c56a2f1bb140e1cde70fe9324952830a86faa6"
        ],
        "Parent": "",
        "Comment": "Imported from -",
        "Created": "2020-09-30T04:14:04.361960576Z",
        "Container": "",
        "ContainerConfig": {
            "Hostname": "",
            "Domainname": "",
            "User": "",
            "AttachStdin": false,
            "AttachStdout": false,
            "AttachStderr": false,
            "Tty": false,
            "OpenStdin": false,
            "StdinOnce": false,
            "Env": null,
            "Cmd": null,
            "Image": "",
            "Volumes": null,
            "WorkingDir": "",
            "Entrypoint": null,
            "OnBuild": null,
            "Labels": null
        },
        "DockerVersion": "19.03.12+azure",
        "Author": "",
        "Config": {
            "Hostname": "",
            "Domainname": "",
            "User": "",
            "AttachStdin": false,
            "AttachStdout": false,
            "AttachStderr": false,
            "Tty": false,
            "OpenStdin": false,
            "StdinOnce": false,
            "Env": null,
            "Cmd": null,
            "Image": "",
            "Volumes": null,
            "WorkingDir": "",
            "Entrypoint": null,
            "OnBuild": null,
            "Labels": null
        },
        "Architecture": "amd64",
        "Os": "linux",
        "Size": 356576580,
        "VirtualSize": 356576580,
        "GraphDriver": {
            "Data": {
                "MergedDir": "/var/lib/docker/overlay2/09ac059db19e82baf759509e08ce979d0cf01276616c5494432c17a279ff66ad/merged",
                "UpperDir": "/var/lib/docker/overlay2/09ac059db19e82baf759509e08ce979d0cf01276616c5494432c17a279ff66ad/diff",
                "WorkDir": "/var/lib/docker/overlay2/09ac059db19e82baf759509e08ce979d0cf01276616c5494432c17a279ff66ad/work"
            },
            "Name": "overlay2"
        },
        "RootFS": {
            "Type": "layers",
            "Layers": [
                "sha256:206a844bb58327e50431b9d36f4605fb9453806df2f2e6a0315b9fb634b6ecde"
            ]
        },
        "Metadata": {
            "LastTagTime": "0001-01-01T00:00:00Z"
        }
    }
]
```
# Steps to Reproduce
whether
`docker pull registry.cn-shanghai.aliyuncs.com/suling/openwrt:rpi3` or
`docker pull sulinggg/openwrt:rpi3` has got the same error.

`standard_init_linux.go:211: exec user process caused "exec format error"`

# Context (Environment)
```
root@DietPi:~# docker info
Server:
 Server Version: 19.03.13
 Storage Driver: overlay2
  Backing Filesystem: extfs
  Supports d_type: true
  Native Overlay Diff: true
 Logging Driver: json-file
 Cgroup Driver: cgroupfs
 Plugins:
  Volume: local
  Network: bridge host ipvlan macvlan null overlay
  Log: awslogs fluentd gcplogs gelf journald json-file local logentries splunk syslog
 Swarm: inactive
 Runtimes: runc
 Default Runtime: runc
 Init Binary: docker-init
 containerd version: 8fba4e9a7d01810a393d5d25a3621dc101981175
 runc version: dc9208a3303feef5b3839f4323d9beb36df0a9dd
 init version: fec3683
 Security Options:
  seccomp
   Profile: default
 Kernel Version: 5.4.51-v7+
 Operating System: Raspbian GNU/Linux 10 (buster)
 OSType: linux
 Architecture: armv7l
```

```
System:    Host: DietPi Kernel: 5.4.51-v7+ armv7l bits: 32 Console: tty 0 Distro: Raspbian GNU/Linux 10 (buster)
Machine:   Type: ARM Device System: Raspberry Pi 3 Model B Plus Rev 1.3 details: BCM2835 rev: a020d3
           serial: <filter>
CPU:       Topology: Quad Core model: ARMv7 v7l variant: cortex-a53 bits: 32 type: MCP family: 7 model-id: N/A
           stepping: 4 microcode: N/A
           Speed: 1400 MHz min/max: 700/1400 MHz Core speeds (MHz): 1: 1400 2: 1400 3: 1400 4: 1400
           Vulnerabilities: No CPU vulnerability/bugs data available.
Graphics:  Device-1: bcm2708-fb driver: bcm2708_fb v: kernel
           Device-2: bcm2835-hdmi driver: N/A
           Display: server: No display server data found. Headless machine? tty: 120x30
           Message: Unable to show advanced data. Required tool glxinfo missing.
Audio:     Device-1: bcm2835-hdmi driver: N/A
Network:   Device-1: Standard Microsystems type: USB driver: lan78xx
           IF: eth0 state: up speed: 1000 Mbps duplex: full mac: <filter>
           IF-ID-1: docker0 state: down mac: <filter>
Drives:    Local Storage: total: 14.84 GiB used: 2.52 GiB (17.0%)
           ID-1: /dev/mmcblk0 model: SC16G size: 14.84 GiB block size: physical: 512 B logical: 512 B
Partition: ID-1: / raw size: 14.58 GiB size: 14.33 GiB (98.25%) used: 2.46 GiB (17.2%) fs: ext4 block size: 4096 B
           dev: /dev/mmcblk0p2
           ID-2: /boot raw size: 256.0 MiB size: 252.0 MiB (98.46%) used: 55.0 MiB (21.8%) fs: vfat
           block size: 512 B dev: /dev/mmcblk0p1
Sensors:   Missing: Required tool sensors not installed. Check --recommends
Info:      Processes: 106 Uptime: 1h 09m Memory: 991.4 MiB used: 191.6 MiB (19.3%) gpu: 16.0 MiB Init: systemd
           runlevel: 5 Shell: bash inxi: 3.0.32
```

SuLingGG, 2020-10-19:
因为 Github Action 提供

SuLingGG, 2020-10-19:
找到原因了...稍晚修复。(手机打字刚刚误关 issues 了抱歉~)

SuLingGG, 2020-10-20:
应该已经解决了。

bookyue, 2020-10-23:
Sorry for the delayed response. It works fine for me.
Thanks for getting this done!

closed 2020-10-23 by bookyue