SuLingGG/OpenWrt-Docker #7: 创建容器后，进入容器一直提示 Container is restarting, wait until the container is running

huangyafei, 2020-11-04:
树莓派 4，之前一直使用的是很老的镜像，大概是一月份时的一个 sulinggg/openwrt:latest，一直使用正常。

今天尝试使用 sulinggg/openwrt:rpi4，把旧的容器删除使用新的镜像创建容器后，使用 `docker exec -it openwrt bash` 进入容器时，一直报错：

```
Error response from daemon: Container c2251ca2bedc3e00f6a1199ed0dfcd27f5cd1b3bacde3c999e8708d778265313 is restarting, wait until the container is running
```

SuLingGG, 2020-11-04:
用

docker logs openwrt

看看容器在启动过程中输出的日志。

huangyafei, 2020-11-04:
> 用
> 
> docker logs openwrt
> 
> 看看容器在启动过程中输出的日志。

一直在报：`standard_init_linux.go:211: exec user process caused "exec format error"`

huangyafei, 2020-11-04:
和 https://github.com/SuLingGG/OpenWrt-Rpi-Docker/issues/6 这个是同样的原因么？

SuLingGG, 2020-11-04:
使用:
uname -a 
查看内核基本信息。

如果提示 arm 或者 armhf 的话，说明你正在使用 32 位内核的系统。

如果提示 aarch64 或者 arm64 的话，说明你正在使用 64 位内核的系统。

即使你是树莓派 4 ，如果你正在使用 32 位内核的系统，你也只能用 rpi2 标签的 32 位 Docker 镜像。如果你使用 64 位内核的系统，那么 rpi3 或者 rpi4 标签的 Docker 镜像才会可用。

huangyafei, 2020-11-04:
armv7l，果然，32 位的，明白原因了，多谢作者：）

closed 2020-11-04 by huangyafei

SuLingGG, 2020-11-04:
嗯嗯，用 rpi2 或者 latest 标签的镜像试试看吧，其实效率差不了多少的~

huangyafei, 2020-11-04:
latest 标签的镜像其实就是对应 rpi2 的对嘛？

SuLingGG, 2020-11-04:
> latest 标签的镜像其实就是对应 rpi2 的对嘛？

是的，为了保证最大程度兼容，latest 标签也指向的是 rpi2 的 32 位镜像，这样树莓派 2~4 都能用。

matthuo333, 2021-04-02:
> 使用:
> uname -a
> 查看内核基本信息。
> 
> 如果提示 arm 或者 armhf 的话，说明你正在使用 32 位内核的系统。
> 
> 如果提示 aarch64 或者 arm64 的话，说明你正在使用 64 位内核的系统。
> 
> 即使你是树莓派 4 ，如果你正在使用 32 位内核的系统，你也只能用 rpi2 标签的 32 位 Docker 镜像。如果你使用 64 位内核的系统，那么 rpi3 或者 rpi4 标签的 64 位 Docker 镜像才会可用。
> 
> 比较典型的，官网下载的 Raspbian 就是 32 位内核 32 位系统。

从Libreelec 官方的github上clone后的项目进行编译，其中对RPi4 也只能选择ARCH=arm , 没有arm64, 知道官方是出于什么考虑么？