SuLingGG/OpenWrt-Docker #8: 或导致树莓派意外重启

saury, 2020-11-12:
树莓派3b+，官方的armv7l，用的 latest tag 的镜像，服务只开了个 v2ray 科学上网。
最近发现树莓派频繁重启，一直疑惑，直到最近运行了一次`docker stop openwrt`，发现不管 stop, restart, kill运行完10秒内必重启。
猜测是哪里出问题会导致docker image 重启或关闭，导致树莓派整机重启。

SuLingGG, 2020-11-12:
Docker 运行 OpenWrt 容器时在进行包含 “杀死 OpenWrt Docker 容器进程” 的操作的时候确实会出现这个问题。目前还不清楚是哪方面的原因。
但是“频繁重启”这个问题我最近没有得到反馈，我建议你试用一下其他系统，比如 ubuntu，或者 OpenFANS 他们的 debian 试试看还会不会出现这样的问题。

sakuyamaij, 2020-12-15:
使用OpenFANS的最新固件也会遇到这个问题。
同时我还遇到一个问题，openwrt固件会每隔6小时重启一次，检查了下日志发现
daemon.info watchcat[3473]: no internet connectivity for 1080 seconds. Reseting when reaching 21600

21600s刚好是6小时，但我实际在这段时间内有在使用网络。难道是watchcat设置有问题

LucienShui, 2020-12-19:
同有这个问题，型号同为 3B+。

SLKun, 2021-01-24:
因为使用了`--privileged`创建容器, 在内部关机或重启时拥有把HOST OS也关机或关机的权限.
事实上, OpenWRT仅给予`--cap-add NET_ADMIN --cap-add NET_RAW`这两个权限即可工作, 如果想要给更多的权限也最好使用`--cap-add ALL`而不是`--privileged`

sonyincn, 2021-02-17:
> 因为使用了`--privileged`创建容器, 在内部关机或重启时拥有把HOST OS也关机或关机的权限
> 事实上, OpenWRT仅给予`--cap-add NET_ADMIN --cap-add NET_RAW`这两个权限即可工作
> 如果想要给更多的权限也最好使用`--cap-add ALL`而不是`--privileged`, 这两者其实也是有区别的

尝试用--cap-add ALL来创建容器，确实能解决这个重启问题，但也会带来其他的问题，op里面的一些参数设置不够权限操作。

fyxw, 2021-03-04:
> 因为使用了`--privileged`创建容器, 在内部关机或重启时拥有把HOST OS也关机或关机的权限
> 事实上, OpenWRT仅给予`--cap-add NET_ADMIN --cap-add NET_RAW`这两个权限即可工作
> 如果想要给更多的权限也最好使用`--cap-add ALL`而不是`--privileged`, 这两者其实也是有区别的

非常感谢，我使用你的建议，用`--cap-add ALL`代替`--privileged`后，在树莓派3B+上的Docker中的openwrt能正确获取到ipv6地址。使用的Lean大佬的源码，编译的armvirt版本

deadlyedge, 2021-03-31:
> > 因为使用了`--privileged`创建容器, 在内部关机或重启时拥有把HOST OS也关机或关机的权限
> > 事实上, OpenWRT仅给予`--cap-add NET_ADMIN --cap-add NET_RAW`这两个权限即可工作
> > 如果想要给更多的权限也最好使用`--cap-add ALL`而不是`--privileged`, 这两者其实也是有区别的
> 
> 尝试用--cap-add ALL来创建容器，确实能解决这个重启问题，但也会带来其他的问题，op里面的一些参数设置不够权限操作。

感谢尝试。能不能大概列举是哪些操作不够权限呢？

closed 2021-09-16 by SuLingGG