SuLingGG/OpenWrt-Docker #10: 有没有人遇到设为旁路由无法上网？

hcq11, 2020-12-18:
分别 装了x86和pi4版本，打开路由界面均正常，手工设置我的电脑网关到此旁路由，发现无法上网，无法正常工作，有人遇到我同样的情况吗？
在路由中诊断发现可以正常ping通外网

SuLingGG, 2020-12-18:
有一些路由器式会出现这样的情况的，多出现在大多数华为路由器，小米的 AX 系列，以及腾达的少数型号。
如果想用 OpenWrt 的话，只有刷固件版 OpenWrt 然后让 OpenWrt 作为硬路由的上级路由这一种方式。（x86 设备需要至少两个网口，树莓派可以通过使用 USB 有线网卡实现）

fang123322, 2021-01-20:
按照教程设置完,没法上网,在HS8545M 光猫测试不行,又加了一个路由在TL-wr886n路由器下也不行

tokyoliyi, 2021-01-28:
这个问题已经解决了~！#15

CodeDreamfy, 2021-04-20:
我也遇到按照教程走完，无法上网的情况，我的硬路由是AX1800，之前一直用的软路由，这次想换成旁路由，结果卡主了不知道怎么解决~

SuLingGG, 2021-04-20:
@CodeDreamfy 据多人反馈，AX 系列对旁路由确实不友好。实在不行就加块网卡组上下级路由吧。

CodeDreamfy, 2021-04-20:
> @CodeDreamfy 据多人反馈，AX 系列对旁路由确实不友好。实在不行就加块网卡组上下级路由吧。

嗯嗯，我再试试，不行了就换回软路由~

CodeDreamfy, 2021-04-21:
> @CodeDreamfy 据多人反馈，AX 系列对旁路由确实不友好。实在不行就加块网卡组上下级路由吧。

我昨天终于搞定了AX1800路由器无法上网的问题，大佬的教程可能还需要添加点，就是在修改OpenWrt的LAN的时候，切换到物理设置，取消勾选：桥接接口。这个选项。
[image]

EnnawYang, 2021-04-21:
被你发现了

CodeDreamfy, 2021-04-21:
> 被你发现了

大佬早知道了？

closed 2021-09-16 by SuLingGG