SuLingGG/OpenWrt-Docker #12: 已知问题

SuLingGG, 2020-12-19:
目前项目内提供的 Docker 镜像存在以下问题:

1. `docker run` 容器后无法启动成功，使用 `docker logs` 命令查看容器日志，返回 `exec format error`:

这个问题大概率出现在 rpi3 和 rpi4 镜像 / 设备上。

出现这个问题的原因在于，你当前使用的是 32 位架构的内核，但是却尝试运行了 64 位架构的 Docker 镜像。

rpi3 和 rpi4 仅适用于宿主机为 64 位架构的内核，如果你当前使用的宿主机镜像使用了 32 位架构的内核，自然不会运行成功。

因为目前官方主推的 Raspberry Pi OS 仍坚持使用 armhf 作为目标架构进行构建 (虽然有推出 aarch64 架构的 Raspberry Pi OS，但本质是 64 位 Debian + Raspberry Pi OS 皮肤)，所以当你在使用 armhf 版本的 Raspberry Pi OS 时，肯定会遇到这个错误。

解决方法:

使用 latest 或 rpi2 标签拉取适用于 32 位内核的 Docker 镜像。

当前镜像的目标架构可以在 **宿主机中** 用命令 `uname -a` 查看:

以 OpenWrt 固件为例：

armhf 架构 (关键字 armv7l)：

```
Linux xxx 5.4.79 #0 SMP Fri Dec 11 02:36:13 2020 armv7l GNU/Linux
```

aarch64 架构 (关键字 aarch64):

```
Linux xxx 5.4.79 #0 SMP Sat Nov 28 13:32:46 2020 aarch64 GNU/Linux
```

2. 在使用 `docker restart` 或 `docker rm -f` 命令重启或强制删除容器时，宿主机会随着容器重启:

此问题可能会在各个镜像出现，目前暂时没有好的解决办法。

3. x86 平台的镜像 (x86_64 / x86_generic) 其他功能一切正常，但在使用 ShadowsocksR Plus + 或者 Passwall 时无法科学上网:

pansila, 2021-01-03:
问个弱鸡问题，初始密码多少？我猜对一次好久不用又忘了

SuLingGG, 2021-01-03:
> 问个弱鸡问题，初始密码多少？我猜对一次好久不用又忘了

用户名: root / 密码: password

WoChen5770, 2021-03-11:
关于 3. x86 平台的镜像 (x86_64 / x86_generic) 其他功能一切正常，但在使用 ShadowsocksR Plus + 或者 Passwall 时无法科学上网:
大佬，能不能让openwrt的镜像自带内核，这样是否可行？

WoChen5770, 2021-03-11:
目前在用ikuai的docker环境，按你的说法，要升级ikuai的内核，目前估计升级不了

SuLingGG, 2021-03-11:
> 关于 3. x86 平台的镜像 (x86_64 /x86_generic) 其他功能一切正常，但在使用 ShadowsocksR Plus + 或者 Passwall 时无法科学上网:
> 大佬，能不能让 openwrt 的镜像自带内核，这样是否可行？

不可行:
<https://www.zhihu.com/question/55143510>

WoChen5770, 2021-03-11:
> > 关于 3. x86 平台的镜像 (x86_64 /x86_generic) 其他功能一切正常，但在使用 ShadowsocksR Plus + 或者 Passwall 时无法科学上网:
> > 大佬，能不能让 openwrt 的镜像自带内核，这样是否可行？
> 
> 不可行:
> https://www.zhihu.com/question/55143510

好的，感谢回复

hyy34, 2023-03-10:
Passwall 关闭DNS选项和里的ChinaDNS-NG 试试。