SuLingGG/OpenWrt-Docker #13: 客户端连接旁路由无法上网

leeyiding, 2020-12-19:
主路由：红米AC2100  
系统：padavan
版本：3.4.3.9-099_20-12-2

旁路由：树莓派4B  
系统：树莓派爱好者基地64位debian
Docker镜像：openwrt:rpi4

openwrt内可正常ping通百度，连接旁路由客户端无网络。
启用passwall 绕过大陆IP模式后，openwrt内可正常ping通百度和google，连接旁路由客户端可正常访问国外网站，但国内网站无法访问。

尝试一下两种方法均无果
1. 主路由padavan中 外部网络(WAN)-外网设置-IPV4硬件加速设置为"Offload TCP/UDP for LAN"
2. 旁路由openwrt防火墙自定义规则添加"iptables -t nat -I POSTROUTING -j MASQUERADE"

请问大佬是否有解决方法，非常感谢！

SuLingGG, 2020-12-20:
按 https://mlapp.cn/1008.html 最后几条方法试一试。
补充:
1.直接关闭主路由的 IPV4 硬件加速和主路由所有相关的加速。
2. 加上这条防火墙规则和删除原有两条规则都试试。或者试试三条规则之间的排列组合
3. Docker 镜像我记得我在构建的时候默认就关闭了 lan 口的桥接，如果没关闭的话关闭 lan 口的桥接，只留 eth0 试试看。

leeyiding, 2020-12-20:
感谢苏老板，我发现防火墙内自带了一条"iptables -t nat -I POSTROUTING -j MASQUERADE"规则，将其注释后，网络恢复正常

closed 2020-12-20 by leeyiding

matthuo333, 2021-04-02:
可以lsmod你的模块么？ 为什么我的PSW不能出海呢？

unniversary, 2021-07-19:
**非常感谢，使用你的方法，我的成功出海！！！！**

unniversary, 2021-07-19:
我的情况是putty进入树莓派后可以ping通百度，但是使用passwall ,无论国内还是国外的网站都ping不通，然后注释掉防火墙那条规则后，马上出海！太棒了

cxwx, 2023-02-19:
docker安装 关闭lan口桥街以后留着 eth0  无法通过页面访问openwrt了，用的x86_64 版本