SuLingGG/OpenWrt-Docker #14: openwrt

wy15, 2021-01-06:


SuLingGG, 2021-01-06:
> 能否再添加一个 openwrt office 版本的镜像

不会考虑了...Docker 版的 Offical 版和 Lean 版再做版本上的区分属实没有太大意义...

wy15, 2021-01-06:
> > 能否再添加一个 openwrt office 版本的镜像
> 
> 不会考虑了...Docker 版的 Offical 版和 Lean 版再做版本上的区分属实没有太大意义...

Lean版的feed感觉比较旧，像新出的xray没有，还有passwall也不是新的，比较关心这两个。

closed 2021-01-06 by wy15

SuLingGG, 2021-01-06:
> > > 能否再添加一个 openwrt office 版本的镜像
> > 
> > 
> > 不会考虑了...Docker 版的 Offical 版和 Lean 版再做版本上的区分属实没有太大意义...
> 
> Lean 版的 feed 感觉比较旧，像新出的 xray 没有，还有 passwall 也不是新的，比较关心这两个。

Lean 版固件 xray 是有的，passwall 也是最新的，不必纠结，具体可以对比版本号。

wy15, 2021-01-06:
> > > > 能否再添加一个 openwrt office 版本的镜像
> > > 
> > > 
> > > 不会考虑了...Docker 版的 Offical 版和 Lean 版再做版本上的区分属实没有太大意义...
> > 
> > 
> > Lean 版的 feed 感觉比较旧，像新出的 xray 没有，还有 passwall 也不是新的，比较关心这两个。
> 
> Lean 版固件 xray 是有的，passwall 也是最新的，不必纠结，具体可以对比版本号。

知道原因了，32位版没有，64位有