SuLingGG/OpenWrt-Docker #16: 希望列出删掉的无线相关包

SLKun, 2021-01-24:
事实上可以通过
```
iw phy ${WLAN_PHY} set netns name ${CONTAINER}
ip netns exec ${CONTAINER} ip link set ${WLAN_IFACE} name wlan0
```
将HOST的无线网卡移到Docker内部的Namespace(需要网卡支持)

所以树莓派板载的无线网卡是可以容器内部的OpenWRT使用的, 希望能列出无线相关包以便于恢复

SuLingGG, 2021-01-25:
我所说的“移除了部分与 WIFI 有关的软件包”指的是相对于 OpenWrt-Rpi 项目而言的。
与 OpenWrt-Rpi 项目相比，此项目只移除了 WIFI 计划等涉及到 WIFI 控制的软件包，WIFI 核心的软件包如 wpad、hostapd 以及板载无线驱动的内核模块等并没有移除 (但 USB 网卡驱动未加入)，您可以用您的方法试一下看板载无线是否可用~

SLKun, 2021-01-25:
基于官方镜像`https://hub.docker.com/r/openwrtorg/rootfs`的板载无线是可用的
但是基于你这边的镜像虽然可以在shell手动使用wifi但是在luci里没有相关的面板

SuLingGG, 2021-01-26:
> 基于官方镜像 `https://hub.docker.com/r/openwrtorg/rootfs` 的板载无线是可用的
> 但是基于你这边的镜像虽然可以在 shell 手动使用 wifi 但是在 luci 里没有相关的面板

这我就不太清楚了，可能和 LuCI 版本有关？
openwrtorg/rootfs 的 LuCI 应该是 LuCI 19，而我的 LuCI 是 Lean 魔改的 LuCI 18。

SLKun, 2021-02-01:
我再试试吧
我之前在官方18.06可以，后来试了下官方19.07好像也不行
总而言之感谢这个项目

closed 2021-02-01 by SLKun

SuLingGG, 2021-02-01:
你如果想实现类似的功能，也可以参考以下两份资料，但是总体来说可能略显复杂：

https://instar.me/archives/858db735.html
https://note.youdao.com/ynoteshare1/index.html?id=2c5c048f77aec55adf5d107d05d4cf79&type=note#/

有道云笔记那篇是群友写的，他把他折腾的过程记录了下来，可以参考一下。

> 我再试试吧
> 我之前在官方 18.06 可以，后来试了下官方 19.07 好像也不行
> 总而言之感谢这个项目