SuLingGG/OpenWrt-Docker #18: 我想把编译好的包导出来，但没能够实现

OnceAce, 2021-01-29:
cd openwrt/bin/targets/$TARGET/$SUBTARGET/
  docker import $DEVICE-generic-rootfs.tar.gz ***/openwrt:x86_64

按照我的理解是一个叫$DEVICE-generic-rootfs.tar.gz的包位于openwrt/bin/targets/$TARGET/$SUBTARGET/路径下，然而我想导出这个包提示没找到，先用cp复制到别处同样提示没找到文件或目录，是我操作不对还是我理解错了哪里，虚心恳求指教。

SuLingGG, 2021-01-29:
对于我的项目，编译生成物可以在图中下方这两个链接中找到。你可以直接下载链接中众多文件中的 RootFS 文件，然后按需求使用。

[image]

OnceAce, 2021-01-30:
> 对于我的项目，编译生成物可以在图中下方这两个链接中找到。你可以直接下载链接中众多文件中的 RootFS 文件，然后按需求使用。
> 
> [image]

首先，感謝大大百忙之中回答問題。

奶牛快傳和另外的有了解過，確實提供了便利。但我還是想問清楚問題所在，加深這方便了解。

“cd openwrt/bin/targets/$TARGET/$SUBTARGET/ ”
”docker import $DEVICE-generic-rootfs.tar.gz ***/openwrt:x86_64  ”
在這行下面我添加了“   cp ./$DEVICE-generic-rootfs.tar.gz /ecc/images/firmware.tar.gz  ”想把文件複製到另外一個目錄下並重命名為firmware.tar.gz  得到的是” cp: cannot create regular file '/ecc/images/firmware.tar.gz': No such file or directory ” 我也嘗試過先複製後mv,但mv出現No such file or directory但複製貌似麼沒出現這問題，我想過幾個可能出現問題的關鍵但是沒法證實，這段時間讓github白消耗資源編譯，於是下決心請教下大大。

ps:上傳到阿里雲出現超時錯誤，不知大大有遇到這情況嗎，也不是每次都是，但占大多數，偶爾會成功幾次。

closed 2021-01-31 by OnceAce

SuLingGG, 2021-01-31:
> > 对于我的项目，编译生成物可以在图中下方这两个链接中找到。你可以直接下载链接中众多文件中的 RootFS 文件，然后按需求使用。
> > [image]
> 
> 首先，感謝大大百忙之中回答問題並提供了方案。
> 
> 上傳到阿里雲出現超時錯誤，不知大大有遇到這情況嗎，也不是每次都是，但占大多數，偶爾會成功幾次。一旦超時，後面的就沒辦法繼續下去，超過6小時了，我想了解下是我個人使用阿里雲docker鏡像容器不對還是網絡問題導致的，要是網絡問題，我想試試按照教程把job默認6小時加大一點。

是网络的问题...Action 机器回国的网络太差了...
所以我还是推荐直接拉 dockerhub 的镜像。如果个人使用完全没必要把镜像推到阿里云的。