SuLingGG/OpenWrt-Docker #19: 关于无法上网的问题【貌似已解决】

huangqingchao, 2021-02-13:
主路由：RedMi AX5
宿主机：树莓派4B
1、按照教程搭建完之后，客户端无法上网，openwrt和树莓派没问题
2、开启 Passwall 后，翻墙网站可以，其它不行
3、关闭 Passwall 后，取消 `桥接模式`，openwrt和客户端都可以ping通baidu
4、打开 Passwall 后，国内外都域名解析不到，ping ip可以ping通
5、自定义防火墙规则，把
`iptables -t nat -I POSTROUTING -j MASQUERADE`
移到最后面并改为
`iptables -t nat -I POSTROUTING -o eth0 -j MASQUERADE`
就可以了

huangqingchao, 2021-02-13:
但是有个问题，就是我的宿主机想要上网或翻墙，但是这时候宿主机访问不到openwrt了，所以怎么办呀

wynemo, 2021-02-14:
https://github.com/SuLingGG/OpenWrt-Rpi-Docker/blob/master/files/etc/firewall.user 我是桥接取消 同时把作者加的规则注释了 再重启下容器 passwall就可以用了。 @huangqingchao 你可以试试

我也是才刚接触openwrt 完全新手 意见仅供参考

huangqingchao, 2021-02-15:
@wynemo 这样设置完，以这个openwrt为网关的客户端，只能上翻墙的网站，国内的不通..

wynemo, 2021-02-15:
[image]
全注释掉 点重启防火墙。 这个插件用的redir之类的吧 不过我没兴趣仔细研究 作者这里主要初衷是想把所有你客户端到openwrt 重定向到openwrt本地的dns 但不知道为啥没生效了。
解决办法 就是你本地客户端设置dns为路由器的ip。
当然那个桥接还是要去掉。
@huangqingchao

huangqingchao, 2021-02-15:
@wynemo 我后面也是这么设置的，没感觉不稳定，挺稳的。就是宿主没法翻墙。因为我的宿主机还需要装些其它应用，需要翻墙，这就不知道如何实现了

wynemo, 2021-02-16:
还以为你说宿主机不能上网。要宿主机翻墙，在宿主机上加路由表应该就可以，试试 `sudo ip route add 0.0.0.0/0 via your_openwrt_ip dev eth0`

broodnes, 2021-05-27:
我的解决方案是在宿主机和容器里面写32位主机路由，让互访流量从其他设备（不能是主路由，并且该设备的网关必须是容器）中转一下

Rainerhu, 2021-09-05:
```
ip link add link eth0 dev eth0.01 type macvlan mode bridge
ifconfig eth0.01 192.168.1.11/24
route add 192.168.1.10 dev eth0.01
route add default gw 192.168.1.10 dev eth0.01
```
我是在树莓派上新建了一个macvlan的虚拟网卡，然后修改路由让所有流量全部通过该网卡指向openwrt的网卡
我openwrt的ip是192.168.1.10 
不能直接用docker里的openwrt的上网的原因可以了解一下macvlan的知识，是因为桥接模式虚拟出来的网卡是无法和父网卡通信的，但是虚拟网卡之间可以通信

closed 2021-09-16 by SuLingGG