SuLingGG/OpenWrt-Docker #20: X64群晖docker安装openwrt ，不能正常使用pw

zazaji, 2021-02-17:
群晖版本DS918+和DS3617都试过，结果都显示 “系统不兼容，终止执行！”。

```bash
2021-02-17 08:12:03:   + [0]Pdnsd (127.0.0.1:7913)...
2021-02-17 08:12:03:   | - [0]上游DNS：8.8.8.8:53
2021-02-17 08:12:03:   - 域名解析：pdnsd + 使用(TCP节点)解析域名...
2021-02-17 08:12:03:   * 请确认上游 DNS 支持 TCP 查询，如非直连地址，确保 TCP 代理打开，并且已经正确转发！
2021-02-17 08:12:03:   - [0]节点列表中的域名(vpsiplist)：192.168.2.1,114.114.114.114
2021-02-17 08:12:03:   - [0]域名白名单(whitelist)：192.168.2.1,114.114.114.114
2021-02-17 08:12:03:   - [1]代理域名表(blacklist)：127.0.0.1#7913
2021-02-17 08:12:03:   - [0]防火墙域名表(gfwlist)：127.0.0.1#7913
2021-02-17 08:12:03:   - 不强制设置默认DNS
2021-02-17 08:12:03: 开始加载防火墙规则...
2021-02-17 08:12:03: 系统不兼容，终止执行！
2021-02-17 08:12:03: 重启 dnsmasq 服务[0]
2021-02-17 08:12:03: 运行完成！
```

WoChen5770, 2021-03-11:
同，提示 系统不兼容，终止执行！

SuLingGG, 2021-03-11:
@WoChen5770 @zazaji 看看历史 issues。

closed 2021-09-16 by SuLingGG