SuLingGG/OpenWrt-Docker #25: 一个建议

ZHonry, 2021-04-24:
首先非常感谢大佬的打包，已经用上这个docker，提出一点个人建议
能否只打包梯子和vpn服务器两种类型服务，因为我相信既然用这个docker做旁路由的，那么应该都是有nas（或者树莓派）的，能否只作为旁路由使用，其他应用功能可以用单独的docker，而不是一个大而全的docker，
如果确实有人需要全功能的可以增加lite版和full版
emmmmm一点拙见大佬见笑了

SuLingGG, 2021-04-24:
首先我为什么一下子集成这么多插件，那就是想最大程度上满足不同人的各个需求。
做精简版这个想法我听了不下 50 次。我只能说做减法比做加法难。

我之前的回复见: https://github.com/SuLingGG/OpenWrt-Rpi/issues/32

维护多个版本是个好主意，但是没办法兼顾到所有人，维护三个版本，那么就有人会要求我在这三个版本基础上增减插件，变成六个版本，维护六个版本，就有人要求出十二个版本。每个人都想满足自己的需求，都想要 Docker 里只集成自己正在用的插件，而自己用不到的插件统统不要，在我这边是实现不了的。

而你说的 `我相信既然用这个 docker 做旁路由的，那么应该都是有 nas（或者树莓派）的` 想法，虽然有一定根据，但是我觉得也不能覆盖所有人。

并且因为一些复杂的原因，让 Docker 版 OpenWrt 做成像 OpenWrt-Mini 那样能自行安装插件的样子目前也不好实现。

综上所述暂时不会考虑你的建议。

Nlqqer, 2021-04-24:
@SuLingGG 说的很对，我觉得要是自己有其他需求的可以自行用ImmortalWrt的源码编译

ZHonry, 2021-04-24:
是我考虑不周，还是感谢大佬的docker

closed 2021-04-24 by ZHonry

fflrichard, 2021-07-17:
> 是我考虑不周，还是感谢大佬的docker

有这么好用的docker还不知足😂