SuLingGG/OpenWrt-Docker #26: sulinggg/openwrt:rpi4，STATUS一直处于Restarting状态

vikYao, 2021-05-23:
pi@raspberrypi:~ $ sudo docker run --restart unless-stopped --name openwrt -d --network macnet0 --privileged sulinggg/openwrt:rpi4 /sbin/init
WARNING: The requested image's platform (linux/arm64) does not match the detected host platform (linux/arm/v7) and no specific platform was requested
36e16b15e9b8d5f0cf7ce59dd96286460ce29c24c18fca0ab7a7edf83f848496
pi@raspberrypi:~ $ sudo docker ps -a
CONTAINER ID   IMAGE                    COMMAND        CREATED          STATUS                         PORTS                                                                                  NAMES
36e16b15e9b8   sulinggg/openwrt:rpi4    "/sbin/init"   32 seconds ago   Restarting (1) 9 seconds ago                                                                                          openwrt

另：
sulinggg/openwrt:latest，运行一切正常

SuLingGG, 2021-05-23:
已知问题中我已做过解释：https://github.com/SuLingGG/OpenWrt-Docker/issues/12

vikYao, 2021-05-23:
也就是说tag:rpi4的openwrt镜像是64位的，而tag:rpi2是32位的。那么问题来了，它们之间的性能会有多大的差距？？

SuLingGG, 2021-05-23:
这个问题与架构没有关系，而是因人而异。
如果你心理上认为它们的差距天差地别，那么它们的性能就天差地别。
如果你心理上认为没有差异，那么它们的性能就没有差异。

closed 2021-09-16 by SuLingGG