SuLingGG/OpenWrt-Docker #46: opkg update错误

ysuolmai, 2022-01-19:
Opkg升级源的第一行是个本地文件夹，不明白啥意思，反正就是找不到文件报错了。剩下的都是签名校验错误，请问大家这是啥原因？

/ # opkg update
Downloading file:///www/snapshots/targets/bcm27xx/bcm2709/packages/Packages.gz
*** Failed to download the package list from file:///www/snapshots/targets/bcm27xx/bcm2709/packages/Packages.gz

Downloading http://openwrt.cc/snapshots/packages/arm_cortex-a7_neon-vfpv4/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading http://openwrt.cc/snapshots/packages/arm_cortex-a7_neon-vfpv4/base/Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading http://openwrt.cc/snapshots/packages/arm_cortex-a7_neon-vfpv4/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading http://openwrt.cc/snapshots/packages/arm_cortex-a7_neon-vfpv4/luci/Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading http://openwrt.cc/snapshots/packages/arm_cortex-a7_neon-vfpv4/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading http://openwrt.cc/snapshots/packages/arm_cortex-a7_neon-vfpv4/packages/Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading http://openwrt.cc/snapshots/packages/arm_cortex-a7_neon-vfpv4/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading http://openwrt.cc/snapshots/packages/arm_cortex-a7_neon-vfpv4/routing/Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading http://openwrt.cc/snapshots/packages/arm_cortex-a7_neon-vfpv4/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading http://openwrt.cc/snapshots/packages/arm_cortex-a7_neon-vfpv4/telephony/Packages.sig
Signature check failed.
Remove wrong Signature file.
Collected errors:
 * copy_file: ///www/snapshots/targets/bcm27xx/bcm2709/packages/Packages.gz: No such file or directory.
 * file_copy: Failed to copy file ///www/snapshots/targets/bcm27xx/bcm2709/packages/Packages.gz to /var/opkg-lists/openwrt_core.

zhiyuanzhai, 2022-01-21:
看起来像是直接锁死了里面的软件。难道是因为怕滚崩？

ysuolmai, 2022-01-23:
> 看起来像是直接锁死了里面的软件。难道是因为怕滚崩？

不知道哇，看起来sulinggg大大几个月没再更新代码了，都不知道还能不能用了 T_____T

ysuolmai, 2022-01-26:
更新：换成新的镜像了，好像可以了

closed 2022-01-26 by ysuolmai