SuLingGG/OpenWrt-Docker #55: 玩客云DOCKER要用哪个版本，试了几个都一直不断重启中

lanyucomputer, 2022-04-11:
97b9743ae7418c9f7f32bec568cc7b5aafe18ade0bb90b745ea698d26b54a460
AppArmorProfile
Args [ ]
Config { AttachStderr: false, AttachStdin: false, AttachStdout: false, Cmd: /sbin/init, Domainname: , Entrypoint: null, Env: PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin, Hostname: 97b9743ae741, Image: sulinggg/openwrt:arm_cortex-a7_neon-vfpv4, Labels: [object Object], OnBuild: null, OpenStdin: true, StdinOnce: false, Tty: true, User: , Volumes: null, WorkingDir: / }
Created 2022-04-11T08:13:19.722810931Z
Driver overlay2
ExecIDs
GraphDriver { Data: [object Object], Name: overlay2 }
HostConfig { AutoRemove: false, Binds: null, BlkioDeviceReadBps: null, BlkioDeviceReadIOps: null, BlkioDeviceWriteBps: null, BlkioDeviceWriteIOps: null, BlkioWeight: 0, BlkioWeightDevice: , CapAdd: null, CapDrop: null, Cgroup: , CgroupParent: , ConsoleSize: 0,0, ContainerIDFile: , CpuCount: 0, CpuPercent: 0, CpuPeriod: 0, CpuQuota: 0, CpuRealtimePeriod: 0, CpuRealtimeRuntime: 0, CpuShares: 0, CpusetCpus: , CpusetMems: , DeviceCgroupRules: null, Devices: , DiskQuota: 0, Dns: , DnsOptions: , DnsSearch: , ExtraHosts: null, GroupAdd: null, IOMaximumBandwidth: 0, IOMaximumIOps: 0, IpcMode: shareable, Isolation: , KernelMemory: 0, Links: null, LogConfig: [object Object], MaskedPaths: null, Memory: 0, MemoryReservation: 0, MemorySwap: 0, MemorySwappiness: null, NanoCpus: 0, NetworkMode: macnet, OomKillDisable: false, OomScoreAdj: 0, PidMode: , PidsLimit: 0, PortBindings: [object Object], Privileged: true, PublishAllPorts: false, ReadonlyPaths: null, ReadonlyRootfs: false, RestartPolicy: [object Object], Runtime: runc, SecurityOpt: label=disable, ShmSize: 67108864, UTSMode: , Ulimits: null, UsernsMode: , VolumeDriver: , VolumesFrom: null }
HostnamePath /var/lib/docker/containers/97b9743ae7418c9f7f32bec568cc7b5aafe18ade0bb90b745ea698d26b54a460/hostname
HostsPath /var/lib/docker/containers/97b9743ae7418c9f7f32bec568cc7b5aafe18ade0bb90b745ea698d26b54a460/hosts
Id 97b9743ae7418c9f7f32bec568cc7b5aafe18ade0bb90b745ea698d26b54a460
Image sha256:2df62c80702079fe23364188f8d16f9a1a0f908b5fb720efcc6be4aa0b093ea1
LogPath /var/lib/docker/containers/97b9743ae7418c9f7f32bec568cc7b5aafe18ade0bb90b745ea698d26b54a460/97b9743ae7418c9f7f32bec568cc7b5aafe18ade0bb90b745ea698d26b54a460-json.log
MountLabel
Mounts [ ]
Name /sunlinggg
NetworkSettings { Bridge: , EndpointID: , Gateway: , GlobalIPv6Address: , GlobalIPv6PrefixLen: 0, HairpinMode: false, IPAddress: , IPPrefixLen: 0, IPv6Gateway: , LinkLocalIPv6Address: , LinkLocalIPv6PrefixLen: 0, MacAddress: , Networks: [object Object], Ports: [object Object], SandboxID: 798eab2edb207efbb2ec062e050b88882d41c51b1c1a7f743ef1a5e2f7b1e9d0, SandboxKey: /var/run/docker/netns/798eab2edb20, SecondaryIPAddresses: null, SecondaryIPv6Addresses: null }
Path /sbin/init
Platform linux
ProcessLabel
ResolvConfPath /var/lib/docker/containers/97b9743ae7418c9f7f32bec568cc7b5aafe18ade0bb90b745ea698d26b54a460/resolv.conf
RestartCount 12
State { Dead: false, Error: , ExitCode: 132, FinishedAt: 2022-04-11T08:16:28.523430583Z, OOMKilled: false, Paused: false, Pid: 0, Restarting: true, Running: true, StartedAt: 2022-04-11T08:16:28.22811557Z, Status: restarting }

lanyucomputer, 2022-04-11:
sulinggg/openwrt:arm_cortex-a7_neon-vfpv4
这个镜像的状态

lanyucomputer, 2022-04-11:
一直重启

lanyucomputer, 2022-04-11:
docker pull sulinggg/openwrt:armv7这个镜像也是一直重启

lanyucomputer, 2022-04-11:
docker run -itd --name=sunlinggg --restart=always --network=macnet --privileged=true sulinggg/openwrt:armv7 /sbin/init

lanyucomputer, 2022-04-11:
docker run -itd --name=sunlinggg --restart=always --network=macnet --privileged=true sulinggg/openwrt:arm_cortex-a5_vfpv4 /sbin/init 用这个版本可以了

aircross, 2022-10-06:
> docker run -itd --name=sunlinggg --restart=always --network=macnet --privileged=true sulinggg/openwrt:arm_cortex-a5_vfpv4 /sbin/init 用这个版本可以了

有用，谢谢先行者的尝试。

lanyucomputer, 2022-10-08:
> > docker run -itd --name=sunlinggg --restart=always --network=macnet --privileged=true sulinggg/openwrt:arm_cortex-a5_vfpv4 /sbin/init 用这个版本可以了
> 
> 有用，谢谢先行者的尝试。

不客气，就是想让同行者少走弯路