SuLingGG/OpenWrt-Docker #73: 求助：绿联DH2600 [macvlan]：failed to create the macvlan port: operation not supported.

Edwardyx, 2022-12-21:
按照：[在Docker 中运行 OpenWrt 旁路网关 https://mlapp.cn/376.html ](https://mlapp.cn/376.html)
SSH进了绿联的UGOS，打开了网卡混杂模式和创建了macvlan。
执行：
`docker run --restart always --name openwrt -d --network macnet --privileged sulinggg/openwrt:x86_64 /sbin/init`
提示：
`docker: Error response from daemon: failed to create the macvlan port: operation not supported.`
请问怎么解决？

[image]
[image]

xiaokexiang, 2023-03-20:
执行`apt install linux-modules-extra-raspi`后重启试试

SilentC1ty, 2023-04-24:
我也是同样的设备一样的问题，有什么解决办法吗，绿联nas应该本身应该就是基于openwrt的